RoboJackets/robocup-software #1137: Direct and indirect kickoff testing plays

CollinAvidano, 2018-06-06:
@jgkamat  What have we been using for the direct and indirect penalty kicks in terms of plays. Woodward and I thought we had and indirect and direct kick play but we cannot find them. If they do not exist we need to create new ones for comp.

Jason27chan, 2018-06-09:
@jgkamat @joshhting pls help

jgkamat, 2018-06-10:
I answered this on irc so I'll paste the answer here:

<jgkamat> jyc5776: see our_free_kick.py, it has stuff for handling indirect/direct inside of it
<jgkamat> the more specialized restarts like corner and goal kicks have to be direct iirc
<jgkamat> search for gs.is_our_direct for us reading if it was direct/indirect
<jgkamat> although we probably should take the shot anyway just in case it hits the goalie and goes in :P
<jgkamat> also please test all the restarts, we never really test them much on the robots but most of the game is spent in restarts :(

Let me know if you still have questions, we should be handling
indirects/direct free kicks currently as is (I think)

Jason Chan writes:

closed 2018-06-11 by chachmu