RoboJackets/robocup-software #1138: Jiggling on the defense line

Jason27chan, 2018-06-06:
Robots on defense are jiggling because they're not allowed in the goal zone. Should be easy to offset robots by a small distance

jgkamat, 2018-06-06:
I think that the fix for #1098 was not fully tested. I think that if we bump
the range a bit more this should go away, but the problem could be deeper.

Jason Chan writes:

Jason27chan, 2018-06-10:
we're finding a problem where robots are path-planning through each other.

joshhting, 2018-06-14:
Apologies for not looking at this sooner, I'll tackle it this afternoon

Jason27chan, 2018-06-14:
Is this still a problem @chachmu?

chachmu, 2018-06-14:
@Jason27chan Probably, I haven't seen it quite as much since we improved motion control and better calibrated vision but we haven't really run the robots enough to say it is gone for sure.

closed 2018-07-11 by Jason27chan