RoboJackets/robocup-software #1139: Path-planning behind the goal

Jason27chan, 2018-06-06:
Sometimes the robots path-plan behind the goal.

jgkamat, 2018-06-06:
Can someone verify if this is allowed by the rules? Our fields may not have enough space but the fields at comp may, and if it is faster it might be easier to just leave it. Otherwise, we can add an obstacle behind the goal.

On June 5, 2018 9:29:37 PM EDT, Jason Chan <notifications@github.com> wrote:
>Sometimes the robots path-plan behind the goal.

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

joshhting, 2018-06-08:
After looking at the 2018 rulebook, there doesn't seem to be any rule that prohibits this behavior during normal gameplay. Does this happen often? If this is a super rare occurence, then if the RRT wants to go behind the goal then we should let it go behind the goal.

jgkamat, 2018-06-09:
Josh Ting writes:

> After looking at the 2018 rulebook, there doesn't seem to be any rule that
> prohibits this behavior during normal gameplay. Does this happen often? If
> this is a super rare occurence, then if the RRT wants to go behind the goal
> then we should let it go behind the goal.

I don't think this will happen very often, and I would guess it would be
advantageous over the alternative. However, I'm not sure if we have room
behind the field (the field might not be perfectly to spec), and vision might
be lost back there as well (the goal might block us if we get too close).

I'm fine with going either way, this is a pretty small (and unlikely) corner
case to bikeshed over :P

JNeiger, 2018-06-09:
Would it be a easy fix if it does cause some problems?

joshhting, 2018-06-09:
@JNeiger yeah, we would just label the area directly behind the goal as an obstacle

closed 2019-08-25 by kylestach