RoboJackets/robocup-software #1147: Keyboard Controls for Manual Control

CollinAvidano, 2018-06-16:
Due to a controller failure during comp testing it was brought up that we should really have keyboard controls for manual control of the robots in soccer.

jgkamat, 2018-06-16:
We need to make sure that the existing shortcuts don't conflict with this.

Examples are:

| s          | stop       |
| left/right | log scroll |

Collin Avidano writes:

Jason27chan, 2018-07-24:
@jgkamat Is there a place where I could find the existing shortcuts?

JNeiger, 2018-07-24:
I don't think we have added anymore since this was updated last year
https://robojackets.github.io/robocup-software/md_doc__keyboard__shortcuts.html

Jason27chan, 2018-07-31:
@jgkamat How do we implement our existing shortcuts? I found a way that [QT handles key presses](http://doc.qt.io/qt-5/qobject.html#installEventFilter), but we should probably implement our shortcuts the same way?

jgkamat, 2018-08-01:
I'm not entirely sure, but I think these are implemented through qtcreator in
the mainwindow.ui file.

For example, here's (I think) the shortcut 'q' from quicksave in
mainwindow.ui:

  <action name="actionQuicksaveRobotLocations">
   <property name="text">
    <string>Quicksave Robot Locations</string>
   </property>
   <property name="shortcut">
    <string>Q</string>
   </property>
  </action>

Jason Chan writes:

ashaw596, 2018-08-02:
Hmm. They would conflict. Maybe make a mode or something. Otherwise jump
move the frame control to alt +

closed 2022-04-22 by kasohrab