RoboJackets/robocup-software #1161: Expose Config Values to Python

JNeiger, 2018-06-19:


jcarn, 2018-06-20:
Bonus points, do this dynamically so new values are automatically added. Not sure if this is possible, but looping through that section of the UI to get values would be a good start

jgkamat, 2018-06-21:
> Bonus points, do this dynamically so new values are automatically added. Not
> sure if this is possible, but looping through that section of the UI to get
> values would be a good start

I'd hope that's mandatory, I don't want that to be bonus points :P

Without that it would be very hard to maintain.

We can probably pass the values through the python bridge thorough a simple interface...

CollinAvidano, 2018-07-17:
Ok so after some research this is in fact already implemented. On python side you can call something like the following
robocup.Configuration.FromRegisteredConfigurables().nameLookup("PathPlanner/Field Edge Obstacle").value
To read any value from the config PROVIDED it has been initialized. Unfortunately I cannot get it to write ConfigDoubles even though the bridge and c++ side methods are there.

ashaw596, 2018-07-17:
Awesome. So its working for ConfigInts and ConfigBools then?

JNeiger, 2018-07-17:
It's unfortunate that ConfigDoubles don't work. That's pretty much all we use lol

JNeiger, 2019-01-18:
I just tried this again and config doubles are working. (Or at least printing out using the code collin provided). Closing unless someone else can replicate.

closed 2019-01-18 by JNeiger