RoboJackets/robocup-software #1165: Add "fakeouts"

jcarn, 2018-06-20:
Now that passing actually works this is something we can work on. When passing for shooting, rotating some random direction, then quickly rotating back towards the target and shooting is a great way to get a defender out of the way of what would have been a difficult shot. This would hopefully get around teams that do a good job staying inbetween the ball and their goal.

Some things we should consider:

1. Don't do this with a clear shot, it would be pointless and waste time
2. Some measure of urgency would be helpful, an OK shot that will quickly become worse as defenders react is another situation where we don't want to try a fake out
3. Make sure we have good control before trying this, quickly rotating and losing the ball in the process is bad

I haven't thought as much about the second part here, but a quick back-and-forth pass (A to B,  B to A, possibly with A setting up a one-touch shot) is another good method of getting the defense out of the way

jgkamat, 2018-06-20:
This depends heavily on how the defense of the enemy is doing. Maybe we could
somehow figure out how "reactive" they are. Our defense and offense were very
reactive (not proactive), and this would be very effective against us.
However, UBC is the opposite and it wouldn't help at all.

Against a proactive defense, we might be better off doing a one-touch pass or
similar than a fakeout (actually call their "bluff").

jcarn, 2018-06-20:
Agreed, definitely have this behavior toggleable in a time out (or better yet detected automatically)

closed 2025-09-04 by N8BWert