RoboJackets/robocup-software #1166: Preferred robot direction as a factor in role assignment for capture

JNeiger, 2018-06-20:
Role assignment picks the robot to capture based on the linear distance between the ball and the robots. If the ball is moving, it is better to pick a robot on the path to move forward into the ball. This way we don't have to chase down the ball and pivot around it

jgkamat, 2018-06-20:
There's a hacky way of doing this on comp right now, which sort of works.
Hoping we find a better way of doing this though (and that we remove the magic
number).

Joseph Neiger writes:

JNeiger, 2018-11-07:
I think the best approach at this moment is to create a custom `cost_func` that is linearly added to the distance cost. We can then tune the knobs using the `position_cost_multiplier` and our own cost function to get the desired behavior. If we need some sort of non-linear combination, we can just remove the `destination_shape` and do everything in the `cost_func` itself. 

If other subbehaviors decide they want this functionality, we can have the conversation later about whether it's worth adding this role_assignment itself.

@jgkamat Thoughts?

JNeiger, 2018-11-08:
Hopefully that description overall was enough to give a general idea of the goal. [Here ](https://robojackets.github.io/robocup-software/t20163.html) is a good description of role assignment and how to do different operations. [Here](https://github.com/RoboJackets/robocup-software/blob/master/soccer/gameplay/role_assignment.py#L28) is how the `cost_func` is defined, so if you set `cost_func`, it'll have to follow this form. [Here](https://github.com/RoboJackets/robocup-software/blob/master/soccer/gameplay/role_assignment.py#L299) is where it actually gets added into the total cost for each robot in each role. Finally, [here](https://github.com/RoboJackets/robocup-software/blob/master/soccer/gameplay/skills/mark.py#L94) is the role requirements in another play that gives the general form.

kylestach, 2019-08-22:
@JNeiger I'm closing this, as it was apparently fixed by capture rewrite. Feel free to reopen it if it should stay open.

closed 2019-08-22 by kylestach