RoboJackets/robocup-software #1168: Add distance to goal zone from given point

JNeiger, 2018-06-20:


ashaw596, 2018-06-20:
What is this for? robocup.Point(x, y).dist_to(Field.TheirGoalLine) should
do it.

jgkamat, 2018-06-20:
All we need to do for this is to convert the goal area from a compositeshape
to a rect. We had to make it a composite shape before as it was not a plain
rectangle.

Now that I think of it, we probably should just add a near function for
composite shapes, that recurses over it's member shapes.

Joseph Neiger writes:

JNeiger, 2018-06-21:
@ashaw596 We want to check if it's within a distance of the defense area, not goal segment. There's a few ways to do it right now, but it should just be one function call ideally. Some things like picking pass points need it

PR #1170 (2018-07-01): Add nearPoint for shapes
Closes #1168 if it actually works
closed 2018-07-01 by jgkamat