RoboJackets/robocup-software #1177: Record strategic game information

joshhting, 2018-07-02:
Developing plays that make more strategic, reactive, and holistic decisions requires extracting information on the game state and the opposing team's capabilities, and making this information readily available for the play system to access. To start, we are interested in gathering the following information:
- Who last touched the ball
- Opponent's chipper capability and whether a chip is in progress
  - Tiger's Autoref
- Opponent's demonstrated max speed

JNeiger, 2018-07-02:
Are we planning to do some sort of simplified chip detection or should I just build that into the ball filter stuff? All of tigers chip detection is in the ball filter module itself

joshhting, 2018-07-02:
Do that in the ball filter code, that makes the most sense to me

JNeiger, 2018-10-21:
Other things to include is...

- PDF of probable robot locations
- Whether the other team can actually kick
- A kick event object from the ball filter