RoboJackets/robocup-software #1184: Path Planner doesn't move in the correct direction at the beginning of the path

JNeiger, 2018-07-14:
When a path planner repeatedly replans the path while the robot is require to do some sort turn early in on the path, it will never actually converge to a solution for some time. Even a small turn such as this will cause problems.

[image]

In some of the worst case scenarios, we will just move the complete opposite way slowly while trying to move to a target location behind us.

Right now we are only replaning occasionally so it's not a obvious problem, but this solution isn't perfect since you may get multiple replans in a row causing the problem anyways.

I think the cause is the first waypoint isn't actually far enough away from our current velocity vector so when do the linear interpolation for the first frame, it's basically a the same speed and direction we are currently going. I think just getting some of the turn happening in the first waypoint will solve this problem.

@ashaw596 Thoughts?

ashaw596, 2018-07-14:
Hmm. Do you have a good way to replicate this behavior for testing?

JNeiger, 2018-07-15:
I don't have any code offhand, but I think just having the robot velocity not being in line with the target point it's trying to move to will cause it. It should just slowly move in the opposite direction of the target. I'll see if I can edit some of the code in #1174 to cause it consistently