RoboJackets/robocup-software #1224: Fix run-sim2play

JNeiger, 2018-11-11:
One instance does not see vision. I believe it is probably a vision port issue, but I haven't really looked into it.

JNeiger, 2018-11-16:
Oh, see [here](https://github.com/RoboJackets/robocup-software/issues/1117#issuecomment-381377152). You have to set the vision port. I'll close this once someone double checks that

jgkamat, 2018-11-16:
There's supposed to be "documentation" for this here too:
https://github.com/RoboJackets/robocup-software/blob/master/makefile#L49-L50

I knew this was working at comp-time at least, but someone should double check.

kylestach, 2018-11-16:
I can confirm that running `make run-sim2play` and setting "Vision multicast address" to `224.5.23.2` in the left-hand settings pane works correctly.

The warning message really gets buried in all of the console output, even though it should be pretty obvious with the big `!!![WARNING]!!!`. Maybe it would draw more attention if it were bolded or in red or something, but it might get annoying to look at after a while. Also, it doesn't look like grsim has any command line flags to control settings unfortunately :cry:. Regardless, this should probably be documented a little bit more explicitly somewhere.

JNeiger, 2018-11-16:
I think there is a file config for grsim. Would it be possible to copy the working version and change the multicast address?

kylestach, 2018-11-17:
I don't think the grsim config includes multicast address config - it only loads physics and geometry constants, as far as I can tell. I can't see any way to change other parameters other than through the GUI. [Here](https://github.com/RoboJackets/grSim/blob/c1e09c9f7e8c14ee5ff39c8a96edb9ff1f36b73f/src/configwidget.cpp#L217)'s where it loads the ini file describing the robot configuration.

Alan-Penkar, 2018-11-18:
This doesn't work for me even after changing the port.  Works for Jason- Jason and Woodward looked at my machine as well and can't see what the problem is.  Both simulators come back with No Vision.

Help?

jgkamat, 2018-11-19:
> I think there is a file config for grsim. Would it be possible to copy the working version and change the multicast address?

The main problem with automating this is that if someone runs it accidentally,
they have "multicast set up correctly", which has the annoying side effect of
showing up for everyone on the same network. Even if you automate this, you
should make sure that the person running this they are potentially messing
with everyone else on the network so they can fix it quickly if someone
complains :P

I wish there was a cleaner solution to all this fwiw :(