RoboJackets/robocup-software #1234: Wall Tactic

Alan-Penkar, 2019-01-23:
Based on what was discussed today with @JNeiger, @Jason27chan, and @bopas2 for adaptive defense - this is the spec I came up with for the wall tactic.  Please feel free to comment or add anything I missed.

Wall Tactic - 
    This tactic should create a wall between the mark_point (usually will be the ball) 
and the defend_point (usually a spot on the goal line)  The robots will line up equally spaced based on the
defender_spacing parameter on a line or curve (depending on wall_curvature) so that they are on the middle of the block line.

Possible enhancements (ignore for now):
-Determine 'optimal' number of defenders for wall
-Alter spacing based on the direction the mark_robot is facing to make more effective use of fewer defenders
 --This would need to consider camera lag and other things to make sure we don't leave open windows that the opponent could easily pass through

Just 1 state for execute wall
'''
@param mark_point: default to ball location
@param mark_robot: if we don't use mark_point (it's set to None), then use mark_robot.pos as the mark point.
@param defend point: default to shot_pt from KickEval (or center of goal?)
@param wall_distance: distance from mark point to center of wall (runs along the block line).  default to the minimum opp distance for restart?
@param wall_curvature: Curvature (1/radius) of the circle that players line up on.  Default to 0, which should be a straight line.  Need to set a minimum depending on wall_distance, number of robots on the wall, and robot_radius, still need to determine details
@param defender_spacing: default to ball radius / 2 for now.
@param num_defenders: default to 3 based on what we discussed today (easily could change)
'''

jgkamat, 2019-01-23:
This sounds very useful. It can also be used in many other places like in
line_up and the defense against direct (and indirect) restarts.

The main problem that will need to be solved is finding a distance between the
robots that they can protect against shots but far enough so they don't
interfere with each other's path planning. (or maybe, the path planner can be
altered to make this cleaner)

kylestach, 2019-08-22:
I believe this was closed by #1240. @bopas2 please re-open this issue if that's not correct.

closed 2019-08-22 by kylestach