RoboJackets/robocup-software #1239: Path planner occasionally doesn't create a continuous path in the velocity state

JNeiger, 2019-02-07:
This is specific with partial paths and re-planning every single frame. I haven't tested anything else.

When there is a path from the previous frame, the first X seconds are taken off and used as the initial part of the next path. This sometimes produces a situation where the second path and the first path have a discontinuity in the velocity state. It seems specific to large changes in the path target where the previous was a straight line path and the second path being appended is a hard turn in any of the directions. 

Here are a sample of saddle points where the first point set is the last pos/vel of the previous path and the second set point is the first pos/vel of the appended path. 

```
Run 1
Point(-1.13272, 1.78457)Point(-0.12078, -0.297487) - Prev path
Point(-1.13272, 1.78457)Point(-0.231728, -0.570758) - Appended path
```
```
Run 2
Point(-1.1357, 1.77774)Point(-0.109906, -0.254372) - Prev path
Point(-1.1357, 1.77774)Point(-0.175222, -0.405543) - Appended path
```

This picture doesn't correspond to the above data, it's from a separate run, but it shows the previous path, the new path, and a later picture of the velocity discontinuity.

[image]
Initial path

[image]
Next frames appended path

[image]
The discontinuity at a later point

Here is another picture of it happening over a larger path. You can see the fast transitions from red to a deep blue.
[image]

[Here](https://github.com/RoboJackets/robocup-software/pull/1180/files?utf8=%E2%9C%93&diff=unified#diff-bfec79fac28b5f67abeb89d135d4b0c8R136) is the previous path snipping.
[Here](https://github.com/RoboJackets/robocup-software/pull/1180/files?utf8=%E2%9C%93&diff=unified#diff-bfec79fac28b5f67abeb89d135d4b0c8R229) is where we create the new path.
[Here](https://github.com/RoboJackets/robocup-software/pull/1180/files?utf8=%E2%9C%93&diff=unified#diff-bfec79fac28b5f67abeb89d135d4b0c8R236) is where we combine the two paths together.
And finally [here](https://github.com/RoboJackets/robocup-software/pull/1180/files?utf8=%E2%9C%93&diff=unified#diff-bfec79fac28b5f67abeb89d135d4b0c8R241) is the returning of the path.

My first thought is that we may be trying to command an impossible path and it's loosing the initial velocity constraint in order to solve it.

JNeiger, 2019-02-07:
I am able to work around this by limiting the change in the path to something more "reasonable". This should definitely be fixed though

ashaw596, 2019-02-07:
If this is using the same path planner, if its impossible to keep the speed up, it just arbitrarily pretends it's starting velocity is lower.

JNeiger, 2019-02-07:
Was that the planned behavior? It would make life much much easier if I changed that for the capture since I'm trying to figure out the best location to capture by comparing the path of the ball to the path of the robot. I can remove a bunch of the noise that is found with the ball velocity estimation if I do this iteratively one step at a time. This velocity jump puts a little bit of a hamper on that.

I might have a bug, but if I skip paths where there is a velocity jump, it sometimes just doesn't update to the correct intersect positions

ashaw596, 2019-02-07:
Well. that was just the Behavior we had to end up with since I couldn't figure out a better way.  It not suppose to happen much depending on how well the path planner continues the path a bit in the same direction, but its hard.

JNeiger, 2019-02-07:
I gotcha, no worries

closed 2019-02-10 by JNeiger