RoboJackets/robocup-software #1245: Expose Kick Detection to Soccer as a whole

JNeiger, 2019-03-06:
It's in the vision filter rn, but we should expose this to make life easy for some of the plays

JNeiger, 2019-03-26:
- Flag for new kick
- Time since last kick
- Flag if kick is still valid (Hasn't changed directions or stopped rolling)
- Current Kick object (Kick Robot, Kick Position, Kick Time, Kick Speed/dir etc)

kylestach, 2019-08-29:
@JNeiger, do you have thoughts or additional details on implementation for this?

JNeiger, 2019-08-29:
I have a few hours tomorrow that I was going to at least get a partial implementation. At worst, it won't compile or be complete, but it'll have enough that someone else can easily pick up. If that doesn't happen I'll write it out fully what needs to happen by Monday.

kylestach, 2019-08-30:
What does the scope of this look like? I was thinking about using it as a new member project.

JNeiger, 2019-08-30:
It's basically building a little logic and tuning the kick director already made in the vision code. Then pulling the metrics we want (described above) into a system state like container. Then building a python wrapper around it for easy access

JNeiger, 2019-08-30:
If you want to do that, I'll describe it much more detailed with specific references to where I was thinking of adding stuff

JNeiger, 2019-08-30:
I'mma edit this throughout the day as I have free time. 

## Description
As we move towards event based and matric based transitions on plays, a nonzero amount rely on the detection of a kick. This feature is already built into the vision filter due to some technical reasons. The goal of this issue is to expose the detection to python as well as build in a few simple utilities for users.

## Requirements

 - Expose kick detection metrics to python
 - Boolean flag that goes true for a single frame when a kick is detected
 - Time since last kick
 - Boolean flag 

## Design

[Link to design document](link)

_(The design phase can be omitted for "small" features, i.e. doable in a single meeting. However, if this is a subfeature of a larger feature with a design document, please link it above.)_

JNeiger, 2019-09-01:
Do you want me to pound it out, or would you like to use it as an intro issue?

kylestach, 2019-09-01:
Thank you so much! I think the level of detail you have should be plenty for an intro issue.