RoboJackets/robocup-software #1300: Per-Line Checkstyle/Pretty

kylestach, 2019-07-23:
It would be very helpful to use a tool like `clang-format-diff` to only run the formatter on changed _lines_ instead of on all changed _files_. Currently the style checker is useless for large PRs like `comp2019` that touch existing files that may or may not already be properly formatted. Unfortunately, I don't think there's an easy way to integrate this into `stylize` because the tool works by piping the output of git diff into `clang-format-diff` - @justbuchanan, do you have any thoughts?

I've got this set up halfway on my fork (not using stylize, and currently the diff doesn't fail CI but it should), and a sample diff for comp2019 is available at https://circleci.com/gh/kylestach/robocup-software/36.

closed 2019-07-25 by kylestach

justbuchanan, 2019-08-01:
Hi, sorry for the slow reply here. `clang-format-diff` is pretty nifty and does a good job for things like this. It would be possible to integrate this into stylize itself, but since clang is the only stylize-supported formatter that is able to work on diffs (AFAIK), it may not be worth setting it up. Feel free to open a feature request on the stylize repo though.

The biggest issue I remember having with checkstyle and formatting was that the version of `clang-format` on circleci was almost always different than the version of on team member's computers, causing diffs. This is something that `clang-format-diff` will help reduce (because less lines get reformatted), but won't fix.

I'd recommend looking into using a specific version of `clang-format` on circleci and encouraging members to use the same version on their machines.

kylestach, 2019-08-01:
I found a wrapper someone wrote for yapf to only format diffs (actually just a modified version of clang-format-diff, which is also just a wrapper) and PR'd it in #1302. It works pretty nicely. 

I'll look into getting the clang-format versioning down, although it should be alright as long as everyone is on 18.04.

justbuchanan, 2019-08-06:
Cool, that's nice that it works for yapf also. I put this on the stylize todo list: https://github.com/justbuchanan/stylize/issues/32. I'm not sure when/if I'll get to it, but it doesn't look too hard to implement and would be a nice feature to have.

PR #1302 (2019-07-25): Per-Line Checkstyle/Formatter
Fixes #1300, although it doesn't use stylize.

An example of the result of failing diff of this PR (for comp2019) is at https://circleci.com/gh/kylestach/robocup-software/52.