RoboJackets/robocup-software #1309: Export kick/pass configurations to constants file

JNeiger, 2019-08-13:
There are many different places around where we setup specific timeouts, errors values and that type of thing. [See here for example](https://github.com/RoboJackets/robocup-software/pull/1263#discussion-diff-311784598R291). It would be really nice to condense all these configs down into a single location under a `Passing` config or a `Shooting` config.

It would be nice to build up the kicker config as a dict inside `constants.py` such that we can do `self.kicker.aim_params = constants.Passing.AimParams` or something to that effect. As more and more plays are added, we can then start tuning the global configs to be better instead of copying and pasting these local configs around. For right now, just `Passing` and `Shooting` should be good enough. If we need more wiggle room on show good of pass we need, we can split passing up into `DirectPass` and `ThroughPass`, where direct will be very clean and exact while through will just yeet it softly in that direction.

We should think about setting up kick speeds into these config options, but that might be outside the scope of this.

kylestach, 2019-08-13:
Realistically, this should be in the actual config XML, right? It would be really nice to get that working python-side.

JNeiger, 2019-08-14:
It's nice to have all config in one place, but the constants.py access from the python is really nice from a stylistic view.

I'm not even sure you can register config values from python right now even if the reading worked so theres both technical and stylistic herdles to cross

kylestach, 2019-08-14:
Yes, that is correct. Currently the configuration system is not usable from Python. That's something that we need to fix regardless.

Are you saying that in Python, using Python-defined constants is more ergonomic than using config variables? I don't doubt that's the case, but it kind of undercuts the point of having a configuration system at all if we just go ahead and ignore it in Python stuff.

kylestach, 2019-08-14:
Unless you're talking purely from a grouping perspective (i.e. the Params object), which I think can still be accomplished without ignoring our existing configuration system.

JNeiger, 2019-08-14:
It's more python-y to not use the config system, but there's ways around it to shorten the ~60 character call to get the config value you want so it's not cumbersome. It's not hard to overcome, but its a medium sized issue to fix everything to make it usable and convenient. It's also not a "pretty" issue to fix and what we have right now isn't that bad since python reloads on the fly so you can tune the exact same way.

Long story short, yes it'll be nice to combine them, but the current solution isn't bad from a usability perspective so it's not a high priority in my mind. But if it gets fixed, that's one less thing on the todo list

closed 2025-09-04 by N8BWert