RoboJackets/robocup-software #1328: Add automated testing franework

kylestach, 2019-09-04:
## Description
A system to automatically run tests for plays. The GUI will allow users to select sets of tests to run. The system will then automatically place robots and the ball in the locations specified by the tests, run the tests, and report the results.

## Requirements

 - A user should be able to specify a list of robot positions and velocities, ball position and velocity, and game state. 
 - Soccer can be launched in “testing mode” via a command-line argument specifying a python script that specifies above data. 
 - Soccer will then proceed through all testing scripts, running each situation until time limit. 

## Design

[Link to design document](https://docs.google.com/document/d/1jem3jhoX-xXPAJe7aBImg3YXuQ82eKvUvw6GmLgNdgg/edit?usp=drivesdk)

kylestach, 2020-03-28:
@chachmu Is this done?

liquidmetal9015, 2020-04-11:
I think this just needs the time limit and the command line launching. Of course we don't really have the "report the results" part as a part of the current system. The current system just sets up the game state and I think there might be some small issues of things that need to be set. Running it in sim2play mode may also not work great. Other than that we need to actually create testing scenarios.

closed 2020-05-26 by kylestach