RoboJackets/robocup-software #1329: (Tracking issue) Refactor Processor into modules

kylestach, 2019-09-08:
# Description
Currently, Processor has a lot of extra logic to iterate over robots and apply modules to each robot, etc. We want to move logic out of processor, so we will be introducing modules. Each module will be run in sequence (Processor will contain a `vector<unique_ptr<Module>>`) and will own a context pointer (so it will know how to operate).

## Requirements

 - [ ] Module base class
 - [ ] Motion control module
 - [ ] Vision filter module
 - [ ] Vision receiver module
 - [ ] Radio module
 - [ ] Gameplay module (trivial)
 - [ ] Joysticks (see #1325)
 - [ ] Referee module (trivial)
 - [ ] Logger module

## Design

[Link to design document](https://docs.google.com/document/d/13LJ-HotI7wjp8G1xqDhcK5y4MAMSmWZyr55kbajW1qw/edit#heading=h.yvwkw0oatlpt)

Also, see #1303 for information about how we plan to pass information between modules.

jpfeltracco, 2019-09-09:
This might be dumb, feel free to ignore 😉. But I got a bit confused by the term "Module" at first and I think others might too in the future. C++ has had "modules" support coming out for a while (should be slated for C++20 now).

Any other terms that might prevent overloading "modules"? "Manager", "handler", "thingythatdoesstuff"?

kylestach, 2019-09-09:
@jpfeltracco that's a fair point - thinking about it, that's going to be one of those things that causes people to curse my name forever post-2020.

I'm not sure about Manager, I don't really think that captures it. `Runnable` might work (or if we really want to go all out on the Processor theme, `Processable` :stuck_out_tongue_winking_eye: ) - although the -able suffix feels sort of Java-y to me.

oswinso, 2019-09-09:
How about.... `Node`? 😋

kylestach, 2019-09-09:
`Subsystem` maybe? A little bit odd but no conflicts with existing terminology...

kylestach, 2019-09-11:
Update: discussed offline, we're going with `Node`.

closed 2020-05-26 by kylestach