RoboJackets/robocup-software #1340: Coordinated Pass Failure to Terminate

liquidmetal9015, 2019-09-29:
## Description
When playing in simulated and real matches, we often end up in a state in which Coordinated pass should have terminated but it has not. 

1. Passing robot has fully lost the ball at any point before the pass was made
2. Passing robot fails to kick the ball, but is unable to get the ball again due to role requirements
3. The ball was intercepted in between the passing robot and the receiving robot.

All of these situations lead to major paralysis of the team.

[image]

[image]

## Steps to reproduce
1. Create one of the situations in which Coordinated pass should be failing. It will happen naturally when running sim2play

Result: Coordinated pass continues to run when neither the passer or the receiver are in a position to possess the ball.

- [x] This bug has been reproduced by someone else

## Additional information

The primary solution to this problem will probably consist of adding robust termination conditions to the various states of coordinated pass. Adding a conditions that look at the rate of approach of the ball to the receiving robot and at the possession of the ball might be good places to start.

liquidmetal9015, 2019-09-29:
There is actually another major aspect to this: When the passing path becomes invalid due to obstacles getting in the way, coordinated pass tries to pass through them. This behavior needs to be fixed as well, probably by terminating coordinated pass. It's possible that coordinated pass needs some kind of overhaul.

liquidmetal9015, 2019-09-29:
To some degree Situational Analysis based play preemption should help with this, as the play that is calling the Coordinated Pass would be terminated once we enter the pileup state. But the general notion is that we should attempt to fix it in the current system, and that Coordinated pass failing should be an expected possible outcome when any of these things happen.

kylestach, 2019-10-25:
Let's get some quality testing of this now that #1355 is merged, and then if we don't see the bad behavior we'll call it good to go.

closed 2019-11-20 by tjones320