RoboJackets/robocup-software #1344: Adaptive Formation/Adaptive Defence need to be split/refactored to work in the Situation Analysis system

liquidmetal9015, 2019-10-02:
#  Description

This is related to #1343 and <issue about defend goal play> in that it would solve those issues, but it would be good to have the split version of adaptive in addition to other plays for those two situations.

We have a play called adaptive formation that has been our best omni-play so far. But now that we are moving to situation analysis based plays we need to convert it into an "attack goal" play with just its offensive components and an "defend goal" play with just its defensive components. There was some work done on an "Adaptive Defense" that Allen, Tanner, and Joe were involved in that should also be integrated into the "defend goal" play.

## Notes about design

Just to reiterate:
The attack goal situation is active when we have the ball (the ball is exclusively in our robots mouth) and we are within shooting range, which is anywhere on our opponents side of the field.
The defend goal situation is basically the inverse, where our opponents have the ball and are in shooting range.

For this particular defend goal play it may be favorable to just keep the default defense as that is how it was designed to operate.

## Requirements 

 - Create a "Defend Goal" play based on adaptive formation and the work done on adaptive defense 
 -  Create a "Attack Goal" play based on adaptive formation

Jason27chan, 2019-11-10:
adaptive defense has been added as a play renamed as winger wall. See [here](https://github.com/RoboJackets/robocup-software/commit/a4d29d1bf7cd7a1ca561fcd5cbfbb98919eda9bc#diff-c8533ebbaa8968909619b94b7c6fc14c)

Jason27chan, 2019-11-10:
As of right now, the adaptive formation play only controls the offensive capabilities, so we can use it as a attack goal situation.

closed 2019-11-19 by Jason27chan