RoboJackets/robocup-software #1348: There are currently no plays for the "Offensive Pileup" or "Defensive Pileup" situations

liquidmetal9015, 2019-10-02:
## Description

Normally you would design a play of whatever situation in a design document, then create an issue for you or someone else to make it. But since we are currently lacking any plays at all for "Offensive and Defensive Pileup" situations, we are creating an issue to create a play or plays for those situations at all. In the longer run we want to have at least a couple of plays for each situation, but for this issue just create a play or plays for "Offensive and Defensive Pileup" using whatever methodology just so we have a baseline for those situations.

Eventually this play will need to interface with the situation analysis module for score determination, but that functionality is still in development so for right now the play will be developed as a general play and that integration will be added in later on. The integration should be pretty seamless with only the score function being affected, and even then a regular score function will still be the default if we decide to disable situation analysis for any reason.

In the new situational plays paradigm, plays will only be active during a specific subset of game play. So you will only need to handle what happens inside that situation and should assume that your play will be artificially terminated when game play leaves that situation. (You may still want to have your play terminate itself on certain triggers however). Below I will describe the situation that this play needs to cover. 

## Pileup Situations:

The pileup situation is fairly unique and it tends to cover the most chaotic segments of game play where the result is unpredictable. Goals are very often scored in the moments following a pile up, especially in division B.

A pile up situation occurs when robots from two separate teams both have the ball in their mouths at the same time, or there are 3 robots from a mixture of teams in very close proximity to the ball at the same time. Basically there has been a "pile up" of robots around the ball. An "Offensive Pileup" is just a pileup where the ball is on our opponents side of the field and a "Defensive Pileup" is a pileup that occurs on our side of the field.

## Existing code and ideas:

We don't really have all that much for this situation so far. In fact this play will be particularly challenging while #1341 remains unsolved, and that problem is a challenging one. In the longer term we will probably want to have custom low level maneuvering for these situations that bypass the path planner to attempt to wrestle the ball out of the pile up. For now we are mostly want to robustly position robots around the pileup to address what comes next. Those positions could be based on some new evaluation as to what the result of the pileup is going to be.

We have defiantly considered creating behaviors that force a pileup on purpose if we think we can win, or want to slow down game play. That is if these types of plays end up working well. 

The reason this is one issue is that naturally the two situations are fairly related. We aren't sure how we want to tackle related plays yet but using an inheritance structure is one possibility. Its also possible that for right now we just develop a single pileup play and use that play in all pileup situations. In fact, that should almost certainly be the starting point for development on this.

## Notes about design:

Currently plays have to control all the robots on the field, this will not always be the case. It is likely that we will expand our library of "latent behaviors" to help deal with this as needed where latent behaviors are tactics and skills designed to handle robots that are not a part of the primary maneuver the play is executing. For right now leaving robots idle is an option although we will need to assign them latent behaviors for better performance, there are a couple projects working on remedying this. Working around the default defense and goalie will also be necessary for the time being. We may decide to change how those work in the future.

Also keep in mind that writing additional tactics and skills is ideal when designing new behavior, as it will allow us to reuse those behaviors in other places.

## Requirements

- Have a reasonably performant play that fulfills the "Offensive Pileup" situation
- Have a reasonably performant play that fulfills the "Defensive Pileup" situation

## Design
[Insert link to design document here once you have created it](link)

hdemusg, 2019-10-08:
Some initial thoughts:   
- we should do different things if it's in our half vs. if it's in their half. 

Offensive (their half) - get in scoring position:
- if we're in direction of the goal, chip shot towards the net
- else, quick rotation to wrest the ball away?

Defensive (our half) - get the ball out of danger:
- chip shot over attacking robot to clear ball away

liquidmetal9015, 2019-10-08:
For reference, here are some examples of the most common pileup situation. There are pileups involving more robots but the most common ones look like these. This situation usually happens when two robots are sandwiching the ball, and often resolves when the ball squishes out between them to either side.

[image]
[image]
[image]
[image]

JNeiger, 2019-10-08:
A slight rotation and a kick might be enough to clear the pileup and get things back into play. If we're in good positions we can knock it out on them too and get the free kick

hdemusg, 2019-10-08:
That would be a good direction to go in. I think for now, I'll start with just getting robots to the side(s) of the ball and be able to collect

liquidmetal9015, 2020-04-20:
While the info here may still be useful in the future, this is completed as described here.

closed 2020-04-20 by liquidmetal9015