RoboJackets/robocup-software #1353: Collecting ball at end of state transition causes ball to avoid ball

HussainGynai, 2019-10-05:
## Description
When a robot is collecting a ball at the end of a transition in a play it set to not avoid the ball. However if the state changes, the robot that collected the ball becomes unassigned and temporarily avoids the ball and backs away before the next state starts. This causes the robot assigned to collect the ball to try and recapture the ball if a state transition occurs.

## Steps to reproduce
1. Run a captures or collects right before and after and a state transition (Adaptive Formation, Distraction)
2. Wait for a robot to try and collect the ball right before a state transition

Result: Robot will capture ball, then back away from the ball as the play switches states, and then recapture

- [ ] This bug has been reproduced by someone else

## Additional information
Perhaps a bigger discussion on the play system and state machines as a whole is in order.

JNeiger, 2019-10-05:
There is a one frame lag between robot roles being created and the robots being assigned. I haven't looked into it too hard for the exact cause, but this is a major structural problem.

I believe what happens is that we create all the composite behaviors and their subbehaviors, spin them. There are no robots assigned at this point so the new skills are not run. Then we assign a robot to the behaviors. The next frame we then spin them again and now there are robots assigned.

liquidmetal9015, 2019-10-06:
Could a solution to #1338 involve also solving this? I think some kind of outside the current play status or flag for the robot currently handling the ball was a suggested solution to that. Something that could both allow communication between offense and defense and survive state changes. We are talking about solutions in a meeting as I type this.

JNeiger, 2019-10-06:
I don't see a clear path between a solution to that and this. This is more of a structural architecture problem due to the order in which we run the play code and do the robot assignment.

JNeiger, 2019-10-06:
One interesting thing to try to fix this would be to run a skill after it has been assigned a robot for the first time instead of waiting until the next frame. That would be the easiest fix, but it might not be pretty at all

kylestach, 2019-10-25:
@JNeiger one frame of latency is less than ideal but I don't think that's the only issue at hand here. It looks like it backs off for multiple frames.

I think the "correct" solution here is to allow plays to have some sort of memory that lasts between states. The issue here isn't even between two plays - it's in the middle of Adaptive - but it will pose an even bigger challenge when we move to situational plays.

JNeiger, 2019-10-25:
There some other issues with the order in which child/parent spins are called such that it takes a frame for "completed" to move up the composite behavior tree. Additionally, complex skills takes a frame to create a new child skill. 

Frame Count:
 1. Defensive Midfielder completes
 2. Adaptive sees this and transfer to passing, creates coordinated pass, which creates pivot kick
 3. Pivot kick creates capture subbehavior
 4. Capture is assigned a robot

I don't think it's worth to build in a "history" into the plays. It's probably better to just build the state machine in a smarted way to deal with these edge cases as we add more and more large trees of behaviors.

closed 2025-09-04 by N8BWert