RoboJackets/robocup-software #1371: Adaptive Formation Processing Hung (Fully Reproducable)

liquidmetal9015, 2019-10-25:
## Description
We have known that there is some sort of non recovering process hung bug with adaptive formation, but at least on my system, I've found a good way to reproduce it at will.

[image]

## Steps to reproduce
1. In sim, start play with no plays selected
2. Click on the adaptive formation check box a couple of times per second, toggling it on and off over and over again. Within a number of toggles, processing should hang. Sometimes it has taken over 100 cycles but I have not failed to reproduce it on my machine on staging or master.

Here I have listed some samples of the number of toggles that it has taken me to reproduce:
Cycles to reproduce = [15, 17, 43, 1, 5, 2, 1, 18, 104, 54, 75]
I can't tell if it's random or correlates with some factor, starting the clicking right after soccer starts seems to get lower numbers, but that could be placebo. I think you might be able to click arbitrarily fast, I haven't really tested. It is very random, or not so random but I don't know what the variable is.

So processing hangs indefinitely, but we can tell that soccer is not only still running, its using up quite a bit of CPU and memory:

[image]

And in fact the amount of memory trends upwards (mostly, I've seen it go down temporarily) until it reaches about +25%: 

[image]

Until after a period of a minute or two, it crashes with a bad alloc error.

[image]

So that was my working theory as I had seen it happen around 3 to 4 times over, but then I had an instance where the memory utilization stayed at 3 to 3.5 percent indefinitely, and just used CPU without ever throwing the bad alloc error even many minutes later. 

[image]

So my gut says we have some kind of runaway recursion/loop somewhere, but I know that some other people have a better hold on what is going on. It's unclear to me to what degree this impacts regular game play, I know that I have seen adaptive formation very occasionally hang but I am unclear if this will affect situational analysis or adaptive formation in general.

I have tested on both staging and master and it reproduces on both.

I have tried over 100 iterations with both basic_122 and two sided attack and failed to reproduce it with those.

(Also note that this leaves grsim running in the background, which might give us new insight into issue #1351)
[image]
 

If or when someone reproduces this please check the box:
- [ ] This bug has been reproduced by someone else

JNeiger, 2019-10-25:
#1208 was the other time this was happening iirc. I know it occasionally pops up here and there, but I'm not quite sure where at yet. All the optimization stuff has limits on run time

liquidmetal9015, 2019-10-27:
Note: At least one other person has failed to reproduce this error. But jay helped profile it, results as follows.

[image]

liquidmetal9015, 2019-10-27:
Another trace with a memory leak caused by running test one-touch. Also Woodward has not been able to reproduce it on his system.
[image]

JNeiger, 2019-10-27:
If it's when you first start adaptive, record the move targets and build a play that moves directly to those pts. If it still kills itself, it's for sure a path planner and may give an easier more reproducible setup

kylestach, 2020-05-26:
@liquidmetal9015 have you run into this at all recently?

closed 2020-05-30 by kylestach