RoboJackets/robocup-software #1382: Adaptive formation crash

liquidmetal9015, 2019-11-20:
## Description
A crash has been observed in adaptive formation, appearing to be due to indexing into return values in passing evaluations. Info from Joe at the bottom.

[image]

## Steps to reproduce
1. Run adaptive formation for a long time
2. Hope it crashes

Result: It crashes

- [ ] This bug has been reproduced by someone else

## Additional information

This is what Joe, knower of things, had to say about it:
"There is now a min_pass_dist argument in eval_best_receive_point
[Link](https://github.com/RoboJackets/robocup-software/commit/562c1e7349b66810a68e49a43e3e35caad9c8cb5)
To prevent this in the future we should do the kick_point=main.ball().pos type thing"

I would hit up @JNeiger about whats going on.

JNeiger, 2019-11-20:
What basically happened was that we added a new argument to the `eval_best_receive_point`. `min_pass_dist` was added. Specifically in [this commit](https://github.com/RoboJackets/robocup-software/commit/562c1e7349b66810a68e49a43e3e35caad9c8cb5#diff-c4100d06ec36b57dbf2cd3caa44aadc7).
Since there's a lot of arguments and we lucked out this time that it actually caused an error. We should change the way it is called. Instead of just calling it in order like `eval_best_receive_point(A, B, C, D, E ...)`, we should change it to be called like `eval_best_receive_point(kick_point=A, ignore_robots=B ...)`

I propose just doing a ctrl-f and changing all locations this is called with this new format

JNeiger, 2019-11-20:
If not, just add a min_pass_dist arg to every call so the calls match the definition

Jason27chan, 2020-02-26:
I don't see this error, but I fixed what Joe was talking about