RoboJackets/robocup-software #1392: Capture performs poorly and unrealistically in simulator, and ball not getting kicked

liquidmetal9015, 2019-11-25:
## Description
When we are running anything in simulator, we notice that our robots are taking a very long time to settle the ball when capturing and show strange behavior when interacting with the ball. If you run sim and run any plays that capture in realistic situations you will see this behavior. 

Getting consistent capture performance in sim will probably be down to two things, changing our capture parameters, and possibly changing simulation characteristics. Right now it looks like we bump the ball out of the way when we touch it (Which happens much less with the damped dribblers), and the ball gets an rather extreme amount of backspin in that tiny moment of contact - to such a degree that it then rolls back into the dribbler, which has not been our experience on the field as the dribbler generally stalls momentarily when the ball hits it.

In addition we can try to tune the parameters in soccer under capture, which consist of a number of multipliers and things that change how we approach the ball, these will need to be tuned in hardware too but right now we need them to be better tuned for the simulator.

An possibly adjacent problem is a problem where our robots will kick the ball, but it will fail to move, possibly indicating a lack of contact between the kicker and the ball, even seemingly when the ball is perfectly in the mouth and is triggering ball sense. This should probably not happen.

## Steps to reproduce
1. Run sim
2. Watch as we struggle to capture a large portion of the time.

## Additional information
@JNeiger and possibly @kylestach or @chachmu might have input on why this is and what needs to be changed.

Getting this fixed is pretty important to getting good testing results in simulator. When we actually get hardware testing working, we should try to match it up as close as possible but

JNeiger, 2019-11-25:
I would suggest getting grsim closer to reality. The friction is super off. I could never find the second parameter so I chose to have the "slowdown" on the long distance rolling correct instead of the correct "slip" grip while dribbling. It's basically like the ball rolling on ice. See #1233 for the math and stuff.

HussainGynai, 2020-01-12:
It appears that the ball not getting kicked situation has only gotten worse. Strangely it only affects some machines and not others. Currently it seems like the ball is captured, and the kicker kicks but the ball doesn't move, so we're not sure if this is a sim issue or a plays issue. We thought it could have been us moving away from the ball for a frame, but we tried turning off avoid ball for all of pivot kick and still doesn't work. So from this we think its a grsim issue and perhaps is the thing @JNeiger said about the params.

liquidmetal9015, 2020-01-21:
So we have figured that the issue is largely related to the grsim.xml file that is saved in your home directory being wrong. Deleting this file will at least get everyone on the same page. Having a way to purge that file automatically upon make clean or some other make target should be part of fixing this ideally. As Joe said, tweaking the ball friction settings in grsim seems to have a large authority over the observed behavior and looking for settings that make rolling and dribbling realistic are what we'd want ideally.

JNeiger, 2020-01-21:
What I did over the summer was record raw ball position data as I manually rolled the ball across the field. This can be seen in the [`friction_estimation`](https://github.com/RoboJackets/robocup-software/compare/friction_estimation) branch. I did save the chopped data as well as the corresponding graphs on that branch. See PR #1233 and robojackets/grSim#5

1. Specially, I held the ball off field, set it down, and pushed it at a varying speeds up the field. I did this across the field and diagonally as well during different data sets.

2. I then chopped down the recording to remove the initial jitter as the filter caught up to the ball speed and removed any data after hitting the far walls.

3. I then applied a simple sqrt(x^2+y^2) to get the total velocity of the ball.

4. I then applied a linear best fit line to get the slope of the ball velocity. This is the rolling friction in m/s^2.

In order to tune grsim, I applied the same procedure to measure the current ball rolling friction. 
1. I used the right click and drag to apply a "rolling" motion to the ball.
2. I then edited `Ball-ground friction` in the grsim config until the measure rolling friction matched that seen on the field. I used `make run-headmore` so I could edit the variable without having to deal with the grsim.xml or the defaults.

Note that this is only the rolling friction. There are two phases of friction in a kick, the first is the sliding friction due to the backspin on the ball from the dribbler, and the second is the rolling friction where it acts more like a wheel rolling on the ground.

In order to get both the rolling and sliding friction estimates, you will have to have a robot dribble and then kick a ball. You will see a significant, instantaneous change in friction at about 70-80% kick speed. A linear best fit line will have to be applied to both friction phases. I honestly could not figure out which second coefficient changed the ball friction so you will have to figure that out and then try to optimize the 2 grsim variables to match the measured friction values.

Note about grSim, the defaults are only used when no `~/.grsim.xml` file exists. To get everyone on the same page you must delete this file and then never touch the grsim configs. The original plan was to throw it into all the normal make commands, but I got caught up in other things