RoboJackets/robocup-software #1406: When running Sim2Play - Both teams appear to use the blue/first teams ball sense results.

liquidmetal9015, 2020-02-05:
## Description
When running GrSim, the yellow team's ball sense lights will light up in soccer in response to the ball being in the mouth of the corresponding blue robot, and the yellow robots will fail to perform accordingly and fail to capture.

## Steps to reproduce
1. Run sim2play
2. Move the ball to the mouth of a blue robot

Result: The yellow robot's ball sense lights up.

- [x] This bug has been reproduced by someone else (Woodward)

## Additional information
@chachmu is looking into this and is saying that the packets are broadcast to different ports, and so cross-talk should not be possible.
It is possible that this has something to do with radio refactoring or changes, it is unknown how long this bug has been going on for as we were using vision for ball sense up until a little while ago and so weren't paying very much attention to the ball sense indicators.

chachmu, 2020-02-05:
As far as I can tell soccer is correctly assigning itself to the right receive ports for each instance of soccer. I think there's a chance that the issue is that grsim is launched by only one of the instances of soccer so maybe something is being launched wrong. That or grsim is just sending incorrect packets. Without some more involved debugging or wiresharking the packets I can't say for sure what is going on.

chachmu, 2020-02-09:
Wireshark confirms that this is an issue with grSim. I'm going to assume for now that it is only an issue with our fork but it looks like grSim always sends blues ball sense to both teams (╯°□°)╯︵ ┻━┻

PR #1410 (2020-02-11): move the grSim submodule to a new commit that fixes ball sense issues
## Description
Update the grSim submodule to a new commit that fixes ball sense issues.

## Associated Issue
Closes #1406 

## Steps to test
### Test Case 1
1. Put ball in blue robots mouth 

Expected result: Only that blue robot shows ball sense

### Test Case 2
1. Put ball in yellow robots mouth 

Expected result: Only that yellow robot shows ball sense
closed 2020-02-11 by chachmu