RoboJackets/robocup-software #1457: Deprecate DOCIF

oswinso, 2020-05-17:
## Description
Ah, [DoCIF](https://github.com/jgkamat/DoCIF), another in-house tool we used to use that is now deprecated and unused. This is described in our [docs](https://robojackets.github.io/robocup-software/md_doc__unit_tests_and_c_i.html). 

It used to rebuild our docker images and run tests and code coverage, but it hasn't been used afaik from like 2018 (which happens to coincide with the last code coverage report that coveralls.io has).

jgkamat, 2020-05-17:
FWIW, this should have all been removed since #1107, I think the docs are just 
outdated. I'm not sure if the coveralls change is related, but iirc it was 
working at least after that change.

oswinso writes:

> ## Description
> Ah, [DoCIF](https://github.com/jgkamat/DoCIF), another in-house tool we used
> to use that is now deprecated and unused. This is described in our [...]

oswinso, 2020-05-18:
Yeah, it's just the docs + `config.docif` file. It's just that it needs to be cleaned up.

closed 2020-10-18 by kylestach