RoboJackets/robocup-software #1513: [TODO] Make trapezoidal motion work correctly when initial velocity > maximum velocity

kylestach, 2020-06-15:
Currently, if the initial velocity is greater than the maximum velocity, trapezoidal motion will give discontinuous paths. This case should be handling by decelerating to maximum speed, moving at maximum speed for as long as is necessary, then decelerating to the target speed. This may require changes in the `shouldFlip` calculation.

Pull request: #1512 

Link to TODO, or file number/line: soccer/planning/tests/TrapezoidalMotionTest.cpp:96

JNeiger, 2020-06-15:
This may interact with collect in a weird way. IIRC, collect sets the path max speed to ballSpeed+DeltaSpeed during the fine approach. We "should" be close to that speed already, but there's no guarantee

kylestach, 2020-06-15:
It doesn't _completely_ break right now, but the trapezoid calculations are definitely off. I have been noticing some occasional odd capture behavior, so I'll put this as a blocker for merging of motion planning.

closed 2025-09-04 by N8BWert